rust-embedded-community/ssd1306 #1: Add to list of drivers

jamwaffles, 2018-02-16:
Open a PR to add this crate to the [list of drivers](https://github.com/japaric/embedded-hal#drivers) when it's ready.

therealprof, 2018-03-23:
I can certainly help with that.

jamwaffles, 2018-03-23:
Thanks! There's also [the weekly driver initiative](https://github.com/rust-lang-nursery/embedded-wg/issues/39). Should it be added to both?

therealprof, 2018-03-23:
The plan is to concentrate all the different sites and places in one so the information is not as scattered. https://github.com/rust-lang-nursery/embedded-wg/issues/68

therealprof, 2018-03-23:
Just let me know when you'd like to have this added. We can already do so now as [WIP] item.

jamwaffles, 2018-03-23:
WIP is fine for now! I don't think we're that far off a 0.1 release in terms of features but there's quite a lot of doc/blog post writing before removing the WIP flag.

jamwaffles, 2018-04-07:
We're [here](https://github.com/rust-embedded/awesome-embedded-rust#wip) now 😄

closed 2018-04-07 by jamwaffles