rust-embedded-community/ssd1306 #6: Fix duplicate buffer in I2C code

jamwaffles, 2018-03-19:
[`send_data()`](https://github.com/jamwaffles/ssd1306/blob/master/src/interface/i2c.rs#L28-L40) should be using less memory. @scowcron's [implementation](https://github.com/EdgewaterDevelopment/rust-ssd1306/blob/master/src/prelude.rs#L27) uses buffered chunks which might be a good way to go. Adding the initial `0x40` to the beginning of the buffer makes using another buffer a necessity at the moment.

scowcron, 2018-03-19:
relevant, I think: https://github.com/japaric/embedded-hal/pull/47

jamwaffles, 2018-03-19:
That looks good. Until that gets merged/implemented though, I think the best way to go is to port over your chunked tx buffer idea. 17 bytes isn't much, even for a uC.

PR #13 (2018-03-20): Use much less memory in I2C interface
Gone from 1025 byte buffer down to 17, although sending data to the display is marginally less efficient. Should be fine.

@scowcron This is from your `rust-ssd1306` crate. Any thoughts?

Fixes #6.
closed 2018-03-20 by jamwaffles