rust-embedded-community/ssd1306 #7: Travis?

jamwaffles, 2018-03-19:
We should have some kind of CI process!

At minimum, right now I think it should:

- Run `rustfmt` and fail if any code changes to keep code clean and consistent
- Run `cargo test` to make sure the examples build*

*<sub>because there aren't any tests at the moment...</sub>

Anything else?

scowcron, 2018-03-19:
Docs would be nice. take a look at `cargo doc-upload`. I did some fiddling with it here:

https://github.com/EdgewaterDevelopment/rust-ssd1306/blob/master/.travis.yml

jamwaffles, 2018-03-19:
I like that. Is there a way to get it to upload to docs.rs? I can't decide whether using docs.rs or GitHub pages is better at the moment.

scowcron, 2018-03-19:
I've never worked with docs.rs, from what I read doc-upload just updates and pushes to your gh-pages branch. the other thing that might be worth looking at is japaric's trust repo, I didn't really dig into it, but it looks kind of nifty.

jamwaffles, 2018-03-19:
Hmm, at first glance it looks like overkill for this crate at the moment at least. Tbf, pushing to a gh-pages branch gives us more control anyway so it's not such a bad thing.

jamwaffles, 2018-03-31:
Closed by #29

closed 2018-03-31 by jamwaffles