rust-embedded-community/ssd1306 #8: Documentation

jamwaffles, 2018-03-19:
Uncomment [this line](https://github.com/jamwaffles/ssd1306/blob/master/src/lib.rs#L3) and add some basic docs for everything.

jamwaffles, 2018-03-19:
Relevant discussion in #7

jamwaffles, 2018-04-08:
Some specific things I've noticed in the [rendered docs](https://jamwaffles.github.io/ssd1306/master/ssd1306/index.html):

* There should be a platform-agnostic example on the docs landing page with a graphics example, probably exercising all bits of the embedded_graphics API (triangle, square, circle, text, image). This will get updated with a following char mode example when #27 is ready. A link to a runnable example demonstrating the same for each display mode would be good too.
* To extend the previous point, there should be an explanation of the concept of display modes; this driver can't be used without choosing one!
* There should be better links between types/other language constructs in the docs
* Readme:
  * The image needs to go on a diet. It's huge at the moment.
  * `graphics` is on by default and no longer optional so should be de-listed
* The re-exports list needs an audit. This is related to #5 so can probably be fixed over there
* Should there be a description of how all the modules fit together?
* [`builder`](https://jamwaffles.github.io/ssd1306/master/ssd1306/builder/index.html) needs a couple of examples. I suggest:
  * A default/basic one that just sets up a display in graphics mode with SPI
  * One that exercises as many of the builder options as possible that changes the display size, rotation and I2C address in I2C mode.
* [`Rotate180`](https://jamwaffles.github.io/ssd1306/master/ssd1306/displayrotation/enum.DisplayRotation.html#variant.Rotate180) doesn't actually flip the display.
* [`interface`](https://jamwaffles.github.io/ssd1306/master/ssd1306/interface/index.html) should mention the [ridiculous](https://github.com/jamwaffles/ssd1306/blob/master/examples/blinky.rs#L45-L57) and [not so ridiculous](https://github.com/jamwaffles/ssd1306/blob/master/examples/blinky_i2c.rs#L40-L41) types required by RTFM apps.
* Each [mode](https://jamwaffles.github.io/ssd1306/master/ssd1306/mode/index.html#reexports) should have an explanation of what it does. Also worth mentioning that `RawMode` shouldn't be directly used and is only a helper for the builder.
* [`borrow_iface_mut`](https://jamwaffles.github.io/ssd1306/master/ssd1306/properties/struct.DisplayProperties.html#method.borrow_iface_mut) could use a short usage example.
* Make sure each module has a more-than-one-line description of what it does, preferably with a code example.

therealprof, 2018-04-08:
> borrow_iface_mut could use a short usage example.

Actually that should go away and the "DisplayProperties" (or whatever the name will end up to be in the end) should provide proper methods to change configuration and send data.

therealprof, 2018-04-08:
Cf. https://github.com/jamwaffles/ssd1306/pull/49

closed 2018-04-23 by jamwaffles