rust-embedded-community/ssd1306 #15: Examples for stm32f30x

jamwaffles, 2018-03-20:
As [mentioned](https://github.com/jamwaffles/ssd1306/pull/13#issuecomment-374764478), we should have a couple of examples (one with and one without graphics perhaps?) using the [`stm32f30x`](https://github.com/japaric/stm32f30x-hal). The examples at the moment use the stm32f103 chip commonly found on "Blue Pill" dev boards so shouldn't be too difficult to port.

therealprof, 2018-03-23:
I'm not too happy about platform specific examples in generic drivers. Not only will that drag in a lot of dependencies to mostly unstable crates which might change at any time but it'll also cause frustration when people try to use the driver but can't figure out how to get past the platform specifics.

jamwaffles, 2018-03-23:
For the record, neither am I. However, I do like the idea of having _working_ examples with the driver because 

1. It makes development much easier, and
2. People who _are_ using the same hardware can get up to speed more quickly. I've seen device-specific stuff used in @japaric's crates, although I don't think he includes the dependencies for them which fixes one of the problems.

I'd like to explore the idea of [Cargo workspaces](https://doc.rust-lang.org/book/second-edition/ch14-03-cargo-workspaces.html). The driver can be in one crate (`ssd1306`) with examples and all their dependencies in the other (`ssd1306-examples` or something). I'm open to better suggestions, but I think that's a good way to go. Thoughts?

therealprof, 2018-03-23:
It sounds like a better idea. I really haven't looked into workspaces though.

I think japaric mostly uses large blocks of example code as in documentation which seems like a great idea but I haven't quite figured out how to work with code in docstrings efficiently.

japaric, 2018-03-26:
> although I don't think he includes the dependencies for them which fixes one of the problems.

For example specific dependencies I use dev-dependencies. These won't be compiled when the driver is compiled as a dependency, but they (dev-dependencies) are required to be published to publish the driver crate.

jamwaffles, 2019-03-09:
There's a growing corpus of examples in the community from HAL/project repositories, so I'm going to close this issue. It doesn't make much sense to have device-specific examples for a generic crate such as this, however I develop it on an STM32F103, so it makes life much easier for me. It should be pretty obvious which bits of the examples are ssd1306-specific, so gleaning usage out of them should still allow them to provide value.

closed 2019-03-09 by jamwaffles