rust-embedded-community/ssd1306 #30: "no method named `draw` found for type"

chocol4te, 2018-03-30:
I am trying to use this library in my project with little success. When I compile my code as an example it works without issue, but in a standalone project I get the following error:

`error[E0599]: no method named ``draw`` found for type ``ssd1306::SSD1306<ssd1306::interface::I2cInterface<hal::i2c::I2c<hal::<unnamed>::I2C1, (hal::gpio::gpiob::PB6<hal::gpio::Alternate<hal::gpio::OpenDrain>>, hal::gpio::gpiob::PB7<hal::gpio::Alternate<hal::gpio::OpenDrain>>)>>>`` in the current scope
  --> src/main.rs:62:10
   |
62 |     disp.draw(Font6x8::render_str("Hello world!", (0, 0)).into_iter());
   |          ^^^^

error[E0599]: no method named ``draw`` found for type ``ssd1306::SSD1306<ssd1306::interface::I2cInterface<hal::i2c::I2c<hal::<unnamed>::I2C1, (hal::gpio::gpiob::PB6<hal::gpio::Alternate<hal::gpio::OpenDrain>>, hal::gpio::gpiob::PB7<hal::gpio::Alternate<hal::gpio::OpenDrain>>)>>>`` in the current scope
  --> src/main.rs:63:10
   |
63 |     disp.draw(Font6x8::render_str("Hello Rust!", (0, 16)).into_iter());
   |          ^^^^
`

Any idea what it could be?

Thanks!

jamwaffles, 2018-03-30:
> ``ssd1306::SSD1306<ssd1306::interface::I2cInterface<hal::i2c::I2c<hal::::I2C1,

Documentation is pretty thin at the moment, sorry about that! Try adding
`--features graphics` to your Cargo command. It will include the
embedded_graphics crate which adds that method.

therealprof, 2018-03-30:
Have I mentioned that having the main graphics driver depend on a feature-gate is not very user-friendly? 😉

jamwaffles, 2018-03-30:
You have indeed 😁 what do you say about enabling the `graphics` feature by default, but leave it in as a feature so it can be disabled? Not sure how useful that is or if I'm just bikeshedding here.

therealprof, 2018-03-31:
Well, as I said, IMHO it should either be always on or we supplement the project with a simpler alternative (i.e. no external dependency, like the option we have on hand from the third contributer 😉) and use that as default. I think it makes sense to have graphics enabled by default but keep it as a feature if someone absolutely wants to get rid of the dependency.

chocol4te, 2018-03-31:
@jamwaffles Thank you so much! Adding 'features = ["graphics"]' under the ssd1306 dependancy in my Cargo.toml fixed it! :)

closed 2018-03-31 by chocol4te