rust-embedded-community/ssd1306 #33: RTFM and I2C

chocol4te, 2018-03-31:
Hello,

I'm very sorry, but I am having trouble using cortex_m_rtfm and an I2C SSD1306. I have it working in a non-RTFM program, and the only examples of using your library with RTFM are with an SPI interface. I keep getting `variant not found` and `proc macro panicked` errors in `app!`.

How would I write the same blinky example using I2C?

Thanks! :)

jamwaffles, 2018-03-31:
I've added a [PR for an I2C RTFM example](https://github.com/jamwaffles/ssd1306/pull/34) which might help you. It's not noted anywhere, but currently this only works with the nightly version of Rust. Are you using nightly to build at the moment?

chocol4te, 2018-03-31:
@jamwaffles Thank you so much! I am using the nightly version and I see where I was going wrong. Thanks again! :)

closed 2018-03-31 by chocol4te

jamwaffles, 2018-03-31:
That's fine! It helps make this crate easier to use for other newcomers :). Can you quickly explain what was wrong and how you fixed it? It might be something to add to some docs.

chocol4te, 2018-03-31:
I was misunderstanding resource definitions in RTFM. I didn't make a `pub type` for the display before trying to use it in `resources`.

commit 526cfb2e (2018-03-31): Add I2C version of blinky example

This addresses #33 with a hopefully helpful example that uses RTFM's
`app!` macro.

Closes #33
PR #34 (2018-03-31): Add I2C version of blinky example
This addresses #33 with a hopefully helpful example that uses RTFM's `app!` macro.

Closes #33
commit c4b415de (2018-03-31): Add I2C version of blinky example

This addresses #33 with a hopefully helpful example that uses RTFM's
`app!` macro.

Closes #33