rust-embedded-community/ssd1306 #41: Shall we hide the interface Error type?

therealprof, 2018-04-06:
Currently each communication interface uses their own interface specific Error type which is passed down all the way to the caller as associated type which requires knowledge about the interface type being used and hence making that knowledge available throughout the driver. However I don't see why the caller would be interested in handling interface (and even HAL implementation) specific errors since that is a lot of work without any particular benefit. In order to get rid of the complexity I would suggest to rewrite the error to a new custom error type.

therealprof, 2018-04-06:
So you mean invent error types instead of just using `()` as Error type? Have anything specific in mind?

jamwaffles, 2018-04-06:
Oh I see, no `()` is good I think. There's not a lot you can get out of an interface in terms of _why_ some data failed to send so a `Result<(), ()>` seems fine to me.

jamwaffles, 2018-04-09:
Closed by #44

closed 2018-04-09 by jamwaffles