rust-embedded-community/ssd1306 #42: Comman::send() is highly inefficient

therealprof, 2018-04-06:
Even the simplest blinky example compiles down to this horrorshow on ARMv7M:

```
File  .text    Size Name
0.0%   0.6%     74B [10 Others]
2.6%  79.4% 10.1KiB ssd1306::command::Command::send
0.2%   6.9%    900B blinky::init
0.1%   3.3%    424B cortex_m_rt::reset_handler
0.1%   2.3%    300B <ssd1306::mode::graphics::GraphicsMode<DI>>::flush
0.0%   1.3%    166B __aeabi_memset4
0.0%   1.2%    152B blinky::draw_square
0.0%   1.1%    142B cortex_m_rtfm::atomic
0.0%   1.0%    134B <ssd1306::mode::graphics::GraphicsMode<DI>>::set_pixel
0.0%   0.8%    100B memset
0.0%   0.7%     96B memcpy
0.0%   0.6%     80B blinky::idle
0.0%   0.2%     30B blinky::main
0.0%   0.1%     10B __aeabi_memset
0.0%   0.1%     10B core::result::unwrap_failed
0.0%   0.1%     10B core::panicking::panic_bounds_check
0.0%   0.1%     10B SVCALL
0.0%   0.1%     10B MEM_MANAGE
0.0%   0.1%     10B DEBUG_MONITOR
0.0%   0.1%     10B PENDSV
0.0%   0.1%     10B USAGE_FAULT
3.3% 100.0% 12.7KiB .text section size, the file size is 387.9KiB
```

Obviously the compiler fails to see through the nature of the call to generate straight forward output.

It's not as close as bad on ARMv6M but a trivial test application easily doubles in size mostly due to that one function:
```
File  .text   Size Name
0.2%  13.5%   320B [35 Others]
0.4%  27.3%   644B ssd1306::command::Command::send
0.2%  15.6%   368B main
0.2%  14.9%   352B <ssd1306::properties::DisplayProperties<DI>>::init_column_mode
0.1%   6.1%   144B <ssd1306::properties::DisplayProperties<DI>>::set_rotation
0.1%   5.1%   120B <ssd1306::interface::i2c::I2cInterface<I2C> as ssd1306::interface::DisplayInterface>::send_data
0.1%   4.2%   100B <stm32f042_hal::i2c::I2c<stm32f042::I2C1, PINS> as embedded_hal::blocking::i2c::Write>::write
0.0%   3.6%    84B <ssd1306::mode::character::CharacterMode<DI> as core::fmt::Write>::write_str
0.0%   2.7%    64B cortex_m_rt::reset_handler
0.0%   1.8%    42B <ssd1306::interface::i2c::I2cInterface<I2C> as ssd1306::interface::DisplayInterface>::send_command
0.0%   0.8%    18B __aeabi_memcpy
0.0%   0.6%    14B __aeabi_memset
0.0%   0.5%    12B __aeabi_memclr
0.0%   0.4%    10B core::result::unwrap_failed
0.0%   0.4%    10B core::panicking::panic_bounds_check
0.0%   0.4%    10B core::slice::slice_index_order_fail
0.0%   0.4%    10B core::slice::slice_index_len_fail
0.0%   0.4%    10B PVD
0.0%   0.4%    10B TIM1_BRK_UP_TRG_COM
0.0%   0.4%    10B DMA_CH4_5_6_7
0.0%   0.4%    10B EXTI0_1
```

therealprof, 2018-04-06:
Oh, I just noticed this actually varies a lot depending on the interface type, the first bloat output above was from a SPI version and here's the same for `examples/blinky_i2c`:

```
File  .text   Size Name
0.0%   2.4%   108B [13 Others]
0.3%  21.8%   976B ssd1306::command::Command::send
0.2%  14.7%   658B blinky_i2c::init
0.2%  13.2%   590B <stm32f103xx_hal::i2c::I2c<stm32f103xx::I2C1, PINS> as embedded_hal::blocking::i2c::Write>::write
0.1%   9.5%   424B cortex_m_rt::reset_handler
0.1%   8.0%   358B <ssd1306::mode::graphics::GraphicsMode<DI>>::flush
0.1%   5.1%   226B <stm32f103xx_hal::i2c::I2c<stm32f103xx::I2C1, PINS>>::init
0.1%   4.2%   186B __aeabi_memcpy4
0.1%   3.7%   166B __aeabi_memset4
0.1%   3.4%   152B blinky_i2c::draw_square
0.0%   3.1%   138B cortex_m_rtfm::atomic
0.0%   3.0%   134B <ssd1306::mode::graphics::GraphicsMode<DI>>::set_pixel
0.0%   2.2%   100B memset
0.0%   2.1%    96B memcpy
0.0%   1.8%    82B blinky_i2c::idle
0.0%   0.7%    30B blinky_i2c::main
0.0%   0.2%    10B __aeabi_memset
0.0%   0.2%    10B core::result::unwrap_failed
0.0%   0.2%    10B core::panicking::panic_bounds_check
0.0%   0.2%    10B core::panicking::panic
0.0%   0.2%    10B SVCALL
1.5% 100.0% 4.4KiB .text section size, the file size is 290.8KiB
```

jamwaffles, 2018-04-06:
Wow that's pretty awful. Is this on 1.26 nightly? Presumably it's the compiler doing some sort of monomorphisation stuff. Any ideas how to solve?

therealprof, 2018-04-06:
This is with rustc 1.26.0-nightly (9c9424de5 2018-03-27)

And no, no idea yet.

jamwaffles, 2018-04-06:
Strange! What command are you using to dump the sizes?

therealprof, 2018-04-06:
https://crates.io/crates/cargo-bloat

therealprof, 2018-04-06:
You have to use it like this: 
```
# xargo bloat --release --example blinky
```

jamwaffles, 2018-04-06:
So, interestingly adding `#[inline(always)]` to `send_command` fixes everything:

Baseline:

```
File  .text    Size Name
0.0%   0.6%     84B [11 Others]
2.6%  78.8% 10.1KiB ssd1306::command::Command::send::hd19ea563d1c03f2f
0.2%   6.2%    820B blinky::init::h4603ccd5e06efe01
0.1%   3.2%    424B cortex_m_rt::reset_handler::hf0bbd8ccb7786ac1
0.1%   2.3%    300B <ssd1306::mode::graphics::GraphicsMode<DI>>::flush::hc0601ddf4a9db009
0.0%   1.4%    186B __aeabi_memcpy4
0.0%   1.3%    166B __aeabi_memset4
0.0%   1.2%    152B blinky::draw_square::h6dc63a457c971b13
0.0%   1.1%    138B cortex_m_rtfm::atomic::h3f9d353c1fdbed85
0.0%   1.0%    134B <ssd1306::mode::graphics::GraphicsMode<DI>>::set_pixel::h28b3a6dc0ec10486
0.0%   0.8%    100B memset
0.0%   0.7%     96B memcpy
0.0%   0.6%     80B blinky::idle::h29b656a25643bb81
0.0%   0.2%     30B blinky::main::hf01e256dd6b36d9f
0.0%   0.1%     10B __aeabi_memset
0.0%   0.1%     10B core::result::unwrap_failed::h06577483ae04a9d4
0.0%   0.1%     10B core::panicking::panic_bounds_check::h41954b572615b0bd
0.0%   0.1%     10B SVCALL
0.0%   0.1%     10B MEM_MANAGE
0.0%   0.1%     10B DEBUG_MONITOR
0.0%   0.1%     10B PENDSV
3.3% 100.0% 12.8KiB .text section size, the file size is 389.9KiB
```

`inline(always)`:

```
File  .text   Size Name
0.0%   1.2%    74B [10 Others]
0.9%  53.5% 3.3KiB blinky::init::h4603ccd5e06efe01
0.4%  20.2% 1.2KiB <ssd1306::mode::graphics::GraphicsMode<DI>>::flush::hc0601ddf4a9db009
0.1%   6.7%   424B cortex_m_rt::reset_handler::hf0bbd8ccb7786ac1
0.1%   2.9%   186B __aeabi_memcpy4
0.0%   2.6%   166B __aeabi_memset4
0.0%   2.4%   152B blinky::draw_square::h6dc63a457c971b13
0.0%   2.2%   138B cortex_m_rtfm::atomic::h3f9d353c1fdbed85
0.0%   2.1%   134B <ssd1306::mode::graphics::GraphicsMode<DI>>::set_pixel::h28b3a6dc0ec10486
0.0%   1.6%   100B memset
0.0%   1.5%    96B memcpy
0.0%   1.3%    80B blinky::idle::h29b656a25643bb81
0.0%   0.5%    30B blinky::main::hf01e256dd6b36d9f
0.0%   0.2%    10B __aeabi_memset
0.0%   0.2%    10B core::result::unwrap_failed::h06577483ae04a9d4
0.0%   0.2%    10B core::panicking::panic_bounds_check::h41954b572615b0bd
0.0%   0.2%    10B SVCALL
0.0%   0.2%    10B MEM_MANAGE
0.0%   0.2%    10B DEBUG_MONITOR
0.0%   0.2%    10B PENDSV
0.0%   0.2%    10B USAGE_FAULT
1.7% 100.0% 6.2KiB .text section size, the file size is 352.9KiB
```

Very odd behaviour; I would've expected the binary size to go up, not down. Hmm.

therealprof, 2018-04-06:
6.2KiB is still a bit on the excessive side of things and it's somewhat funny that `flush()` just made a huge jump up.

I also played with the GPIO stuff of the SPI interface a bit and it seems that simply removing the `self.dc.set_high();` makes an interesting difference...

jamwaffles, 2018-04-06:
Ah, I thought it would've inlined into `flush()` which gets _called by_ `init()`. Either way yeah the problem is in `command.rs`.

PR #53 (2018-04-16): Changed send_command() -> send_commands() to batch commands
This allows us to send multiple commands at once both speeding up the
communication by quite a bit and cutting down the program size
considerably. Fixes #42.
closed 2018-04-16 by jamwaffles