rust-embedded-community/ssd1306 #70: Re: latest build errors

chocol4te, 2018-05-24:
I'm having the exact same issue with a lot of my projects. It's due to crates using an outdated version of `proc-macro2`. Most use 0.3.8 but the issues are fixed on the latest.

jamwaffles, 2018-05-24:
Yeah, I see the failed daily build all the time :). I had a look at fixing the cortex-m-rtfm-macros crate which is where I think the error lies but gave up. I'm kinda just sitting around waiting for @japaric to fix it ;)

chocol4te, 2018-06-07:
I've made some PRs which should fix everything once merged. :)

closed 2018-06-07 by chocol4te

jamwaffles, 2018-06-07:
[This one](https://github.com/japaric/cortex-m-rtfm/pull/80) and [this one](https://github.com/japaric/rtfm-syntax/pull/8)? Can't wait! Would be great to have this driver building again.