rust-embedded-community/ssd1306 #76: Use with a ssd1305 chip, and display size 132x64.

everett1992, 2019-03-09:
I have [this display](https://www.adafruit.com/product/2675) from adafruit. It's a ssd1305 but in adafruit's circuit python tutorials they use a ssd1306 driver. I ran out of memory programming the chip in python so I'm switching to rust and found this library.

In python I had to define the display size as [132x64](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/issues/24), despite being advertised as 128x32. With this display size the top left pixel is 32x4. If I use a display size of 128x32 then every other line is disabled (looks like an old crt).

I see the same behavior with your library, if I use a TerminalMode driver with DisplaySize::Display128x32 then every other line is disabled. If I use 128x64 the display is much better, each character looks correct, but the text starts off screen. The last columns of the display also do not work, it will show random pixels when the display is initialized and they will not change or clear.

 - Would it make sense to add `Display132x64`?
 - I saw `display.properties.set_display_area` but it isn't public, is there a way to set that?

everett1992, 2019-03-09:
I closed the PR because I think it's too heavy weight right now.

I reverted all changes in the PR back to master. 
Here is my display code, and photo of the display using the crate. 
```rust
 let mut disp: GraphicsMode<_> = Builder::new()                              
         .with_size(DisplaySize::Display128x32)                                                                     
         .connect_spi(spi, dc)                                              
         .into();                                                           
                                                                            
    disp.reset(&mut rst, &mut delay);                                       
    disp.init().unwrap();                                                   
    disp.clear();                                                           
                                                                            
    disp.draw(                                                              
        Rect::new(Coord::new(0,0), Coord::new(127,31))                      
        .with_stroke(Some(1u8.into()))                                      
        .with_fill(Some(0u8.into()))                                        
        .into_iter()                                                        
    );                                                                                                     
                                                                                
    disp.flush().unwrap();                                                      
    loop {}              
```

![Display128x32](https://lh3.googleusercontent.com/mkl4nj8rkOEoEZLSl1EsO4vDbMsR16ZVK1dS5hMmVashRqIaSIhxMYwag5OVNNPwLn4zH58XUQmExo6Xgbiv9IxzGkpgMPvcF3SuotOHPTlBT_BJkVG1ngLgpKkgadvalrdLc6dVEXMA1P5IlOWCNyUJQsGmlynpULEXh9U-jivUPuXMDzF_JRbxz-5DYInbkLhIoG0Bx_QxeCG_n70QHtPQXNtHYTAfommfbzRshR3rJVNC-T_X13wULVWgW0yc6vkfyYd2ht-c-sx8-CUMbAWW1a8d5CLbW8UHaGHwnj1NenvPHYkaGY_RKuG6X1_bu7Zr0oe3WY3wyM2ZisC1oIdYLJiyj4LPdJIdcPvc2ph85SgANQP-k2QR1pg8a3hg4sxhqjAm8_HUyTujXFxQbb398-LbJ-9VkMkfbh5q2PGS4ZS7ZWyG5qdf2GZ-Vqt0w0iAhpD1pRXtMnCTVj7urOWUtTQMiMW8L9il-AkYyeVkHjfeS6-G7a--WlP0f11kp0A2OcyNzIMNiHZ1Ca5sRnjczTymBJMhuaWe0f8jNVYOnHcd-NJAgzJe4854xpZHExHSRBBvPBl4CHucVnvvUA6QEhFg-1AqfGmrzjSCbm-3lB1MQZkxrxzLiJYBaRiqfRGEX9OElBZkOORhHzl_SiILWQT1dRc=w732-h975-no)

If I enable scanline on Display128x32 the scanline issue does go away but the rectangle is offset, it looks like it starts left of the screen. (0,0) is in the top-left direction, TerminalMode supports this.

```diff
-            DisplaySize::Display128x32 => Command::ComPinConfig(false, false).send(&mut self.iface),
+            DisplaySize::Display128x32 => Command::ComPinConfig(true, false).send(&mut self.iface)
```

![scanline fixed](https://lh3.googleusercontent.com/9aJOelc69BsmGOlAk3ZzovaHIvDQyQ8LomorfDI4utD3gAT0nWlmfzPpqpOpLZe9Ct3BG1ehmbHHsWX1GUHN7gKjvVdIWgVb8rJBSj549BMT4O9BdK3k3E1laFJfg4nJGPMLZ0jDTgU7NxSDHiK89VekCMCyIlZR5d8a1Aps8dpnCoWqjQkHs3Nti-3-sDyuHuAynnkycOmAChPt2D57n-mjwjfMJWSVWd1XdUiYbU73mAjgRMYQv2Arn85MqAKGwCA5x8A8n5z5jnen2sAxx52HdMAxej9toEhwRxEL0Pn-ViE1tl7EWXPFnd43apvQlG9pSsTM-GgFZy85n7lUE5gLOC1vgv6B8SL2-jtEt0EE1wbXiYOdi2sUnDHfWiSCKcBw3EPTgi3iTXyc-g_2y3UIJVH8755_17zvftFzhdtzqmfXlYoNM0yWJ3BJ_EdvZHRPhRCeiqECwC0c1UjBlkFbCYa4SNJYeWCbJN0pNITAJ5LnZXPTnaP6Xd21-1K50uFXFT3tf9qRr___LY-M8YCWKTi_wwo4-UaHKfQcByib_uS3Ho_HE8LwXW-4G_C7FGBG0vWt8zxhoGkTeFbAXfIB_mwx6VUBGUhRfmy7tpyXZbICUerpGzubMc4jbU9w6i0qKSY-D4Vka04Q7ZAEUsgZhd1fkos=w732-h975-no)

![terminal mode](https://lh3.googleusercontent.com/8T4MrMcAhioaov6v1od3JehMkGVui3KeIfVsKGvScGXyVcR80FTZcUs-DfT-PJqjhJr9iBjliNTeBs9fnLJpjAEN5jxSh1DnB6UTvbxytF8LHhhiGyzGtAkavFr60gNGS8Hq71Adoz_ST7rCGIOp5Ki4ribepDrld8J3WcoRkaWevbkPiL3AUB0-6TFjC3qHHWMvLPTf_2_50eJIW1hsgq-N20F7TihfC9loJdNi3lOH5x5SqU9JGGV5dWoY_th6A3u8tlXARRHHfnhZ6noXZ480ggbeKLiW1EJQ5DkATOmQfqQa8lEUUxd0TAgK4S3nOM7m3ZzBHQr1zLf9GxPIO7ft-2iojI8yWUCyrCFAImL1pNSTAytrcXVQfFKQkkw5f-SE-iX0jRja5xuuDs09iRFY2LGBtZo0cxevt9YJg6woNW5fS9F8-3KFrz1taVzuQoxVwdOZqyHkk90pYSQsa_kWH1z_h43Z4q18m-J-2xLnHdwnQF6abtDTX_Eloc3HbIRzWJfdGPG8QJsbjTjfyLRTmYMjUH-tnPijXjhgVIyywSM0juJjJQmTCV5lddWke4_87LDW6HFqYXMcJkQFQNCKgwYVSlhmCsK2QwGjZl6V-HOtFkOtTawO9CaQfCOzH5MO-gD8lzrfSyI0kQsVStJuJA1PHqg=w732-h975-no)

jamwaffles, 2019-07-01:
Hi, just bumping this old issue. Did you ever get any further in your investigations?

everett1992, 2019-07-05:
I never solved the offset issue.

closed 2019-09-02 by jamwaffles