rust-embedded-community/ssd1306 #84: Feature request: display power off

tpwrules, 2019-07-01:
Is it possible to power off the display using command 0xAE? This would enable saving power and OLED life in long-running applications.

jamwaffles, 2019-07-01:
Certainly! Would you like to have a go at a PR? If not I'm happy to pick this up as I've got the hardware on hand to test.

tpwrules, 2019-07-01:
Yeah, I'll give it a try. I do too but the library has been very useful for me so I'd like to be useful back.

jamwaffles, 2019-07-01:
I think it would be better done as a new method because the programmer has the flexibility to call `.release()` separately or not. If the interface should be released, one can call `.display_off().release()` (or whatever method name you choose), otherwise just `.display_off()` and `.display_on()`. Does that make sense for your use case? I'm open to naming suggestions FWIW :slightly_smiling_face:.

PR #86 (2019-07-02): Implement switching display power
Implements #84 . Tested using [Adafruit OLED Featherwing](https://www.adafruit.com/product/2900). Display has no issues accepting commands while switched off so I don't believe there is a need to implement any interlocks or new state.

Closes #84
closed 2019-07-02 by jamwaffles