rust-embedded-community/ssd1306 #85: TerminalMode 90 and 270 degree rotations incorrect

jamwaffles, 2019-07-01:
As reported in #47.

In `TerminalMode`, the text isn't rotated properly in these two rotations. Instead, it's just drawn backwards for 90 degrees:

[image]

And like this for 270 degrees:

[image]

0 and 180 degrees work fine.

bugadani, 2020-06-12:
Are graphics mode rotations correct? As far as I can tell by looking at the code, only the x and y coordinates are flipped, so I think the displayed image is horizontally flipped?

jamwaffles, 2020-08-03:
Closed by #129 I believe.

closed 2020-08-03 by jamwaffles