rust-embedded-community/ssd1306 #87: Builder examples are not tested and don't work

TDHolmes, 2019-07-21:
Looking at the documentation for the Builder module, it states I can do the following:

```
use ssd1306::prelude::*;
use ssd1306::Builder;

fn main() {
    let mut disp: GraphicsMode<_> = Builder::new()
        .connect_i2c(i2c)
        .with_size(DisplaySize::Display128x32)
        .into();
}
```

However, I get the following error:
```
error[E0599]: no method named `with_size` found for type `ssd1306::mode::displaymode::DisplayMode<ssd1306::mode::raw::RawMode<ssd1306::interface::i2c::I2cInterface<hal::sercom::I2CMaster3>>>` in the current scope
  --> src/main.rs:49:10
   |
49 |         .with_size(DisplaySize::Display128x32)
   |          ^^^^^^^^^
```

The documentation should be fixed and enable testing on them so this doesn't happen again

TDHolmes, 2019-07-21:
Oh I think I see the issue - looks like you have to connect the comms as the last step. However, I think the issue of them not being tested is still valid - adding some hidden boilerplate will enable these examples to be tested.

Also, documenting that the connect step has to come last would be good.

jamwaffles, 2019-07-21:
Ouch, sorry that you found the example confusing! I'll take a look at why they're not tested and I'll add some docs in to clarify the ordering. Thanks for pointing these issues out!

TDHolmes, 2019-07-21:
No worries, I'm liking the driver now that I have something working! This and #88 had me confused for a bit though haha.

jamwaffles, 2019-07-21:
Fixed and released in [0.2.6](https://crates.io/crates/ssd1306/0.2.6). Thanks for raising this issue!

PR #89 (2019-07-21): Clarify builder docs, test doc examples in CI
* Clarify builder docs as per #87 
* Test code examples in CI (also #87)
* Check doc links using `linkchecker` in CI

Closes #87
closed 2019-07-21 by jamwaffles