rust-embedded-community/ssd1306 #88: `.draw()` methods seem to not work with embedded-graphics v0.5+

TDHolmes, 2019-07-21:
I've been able to draw fonts and such with embedded-graphics v0.4.5 with this code:
```
use embedded_graphics::prelude::*;
use embedded_graphics::fonts;

use ssd1306::prelude::*;
use ssd1306::Builder;

#[entry]
fn main() -> ! {

    // ...snip...

    let mut disp: GraphicsMode<_> = Builder::new()
        .with_size(DisplaySize::Display128x32)
        .connect_i2c(i2c)
        .into();

    disp.init().unwrap();
    disp.flush().unwrap();

    disp.draw(
        fonts::Font12x16::render_str("Hello Rust!")
            .into_iter(),
    );
    disp.flush().unwrap();
```

but once I update to latest embedded-graphics, I get the following error:
```
error[E0599]: no method named `draw` found for type `ssd1306::mode::graphics::GraphicsMode<ssd1306::interface::i2c::I2cInterface<hal::sercom::I2CMaster3>>` in the current scope
  --> src/main.rs:55:10
   |
55 |     disp.draw(
   |          ^^^^
   |
   = help: items from traits can only be used if the trait is in scope
help: the following trait is implemented but not in scope, perhaps add a `use` for it:
   |
14 | use embedded_graphics::Drawing;
   |
```

Is this expected? I can't easily see why this is breaking in the first place. Explicitly calling `use embedded_graphics::Drawing` doesn't seem to fix it for me.

jamwaffles, 2019-07-21:
I think it might be due to the version mismatch between the SSD1306 crate (`0.4.x`) and the latest embedded-graphics release (`0.5.x`). Using `0.4.x` for now should fix this issue. I'll be upgrading the embedded-graphics dependency when I have a spare moment 🙂

TDHolmes, 2019-08-26:
FYI, still seeing this error even if I use your branch from #90. Not sure if you expect that or not... 😬

jamwaffles, 2019-08-29:
I checked out the branch in #90 and it compiled the examples in this crate successfully. Do you have a project you could link to so I can take a look? Are you using the same version of embedded-graphics in your `Cargo.toml` as in the `ssd1306` crate?

TDHolmes, 2019-09-01:
Alright, now I'm very confused... I can compile just fine with the following `Cargo.toml`:
```
ssd1306 = { git = "https://github.com/jamwaffles/ssd1306.git", branch = "embedded-graphics-0.6" }
embedded-graphics = "0.6.0-alpha.1"
```

but when I check out `embedded-graphics` myself and point my dependency to my local version, even if it's checked out to the identical commit as what is published for a given version, it fails with the above error. 

```
ssd1306 = { git = "https://github.com/jamwaffles/ssd1306.git", branch = "embedded-graphics-0.6" }
embedded-graphics = { path = "../embedded-graphics/embedded-graphics" }
```

Any ideas why that would be the case?

jamwaffles, 2019-09-02:
I think your first example works because the versions are identical. My hunch is that your second version doesn't work because the `embedded-graphics-0.6` branch uses version `0.6.0-alpha-1` and the `path` property in your `embedded-graphics` dependency doesn't match up with that, even if it's the same Git commit - Cargo won't be able to figure out that they're the same code.

Using your first snippet seems ok to me for now as an interim solution. I'm going to release embedded-graphics `0.6.0-alpha.2` in a mo which can get a matching alpha release of `ssd1306`. I think this should mean you can have your simple Cargo dependencies back:

```toml
ssd1306 = "0.3.0-alpha.1"
embedded-graphics = "0.3.0-alpha.2"
```

TDHolmes, 2019-09-03:
I think you're correct. Once I used a local copy of `ssd1306` that also pointed to my local copy it seems to be working again.

jamwaffles, 2019-09-04:
I've just released `ssd1306` `0.3.0-alpha.1`. This uses `embedded-graphics` `0.6.0-alpha.2` which should hopefully fix your problem, provided you use e-g `0.6.0-alpha.2` in your project's `Cargo.toml`. Please let me know how you get on and I'll close this issue if your problem is resolved.

jamwaffles, 2019-09-10:
No news must mean good news ;). Closing as resolved, but please re-open if the issue persists.

closed 2019-09-10 by jamwaffles