rust-embedded-community/ssd1306 #102: Fail to compile

invis87, 2020-02-03:
- Version of `ssd1306` in use (if applicable): master branch (or "v0.3.0-alpha.2")

## Description of the problem/feature request/other
Fail to compile this project because `embedded-graphics` removed `Drawing` trait. 
```
cargo build  --example image_i2c
   Compiling ssd1306 v0.3.0-alpha.2 (/Users/pronvis/rust/xxx/ssd1306)
error[E0432]: unresolved import `self::embedded_graphics::Drawing`
   --> src/mode/graphics.rs:267:5
    |
267 |     Drawing,
    |     ^^^^^^^ no `Drawing` in the root
```

jamwaffles, 2020-02-03:
Thanks for reporting this. It's now fixed and released in [0.3.0-alpha.3](https://crates.io/crates/ssd1306/0.3.0-alpha.3).

closed 2020-02-03 by jamwaffles

invis87, 2020-02-03:
Wow... interesting fix))) Thanks!

jamwaffles, 2020-02-04:
No problem! Apologies for the breakage