rust-embedded-community/ssd1306 #122: Incorrect min_y?

bugadani, 2020-06-06:
Based on [this metric](http://reviewthecode.blogspot.com/2016/01/wtf-per-minute-actual-measurement-for.html) I believe the linked line may be incorrect.

https://github.com/jamwaffles/ssd1306/blob/96170fd21419e73fdb2299f395b07836d52d090f/src/mode/graphics.rs#L161

If that's supposed to be height, I'm happy to fix it, otherwise please provide an explanation why it's width.

jamwaffles, 2020-06-07:
Yeah that definitely looks like it should be `height - 1`. There's a similar block of code [here](https://github.com/jamwaffles/ssd1306/blob/96170fd21419e73fdb2299f395b07836d52d090f/src/mode/graphics.rs#L111-L114), so it might be worth moving the min/max reset into a private method.

bugadani, 2020-06-07:
I haven't given it much thought, but shouldn't these be rotation aware values? Is min_y the same when rotated by 90 or 270 degrees?

therealprof, 2020-06-07:
Yes, there're the other variables which take the rotation into account. However it's a bit weird to have the manual comparison in the code. There're `max()` and `min()` functions for that.

bugadani, 2020-06-07:
> There's a similar block of code [here](https://github.com/jamwaffles/ssd1306/blob/96170fd21419e73fdb2299f395b07836d52d090f/src/mode/graphics.rs#L111-L114), so it might be worth moving the min/max reset into a private method.

You mislead me :) That block is marking the whole display to be updated, the original block resets the area to nothing.

jamwaffles, 2020-06-08:
Oop yes you're right, min/max are swapped there. My bad!

PR #125 (2020-06-23): Type-level display size
Hi! Thank you for helping out with SSD1306 development! Please:

- [ ] Check that you've added documentation to any new methods
- [ ] Rebase from `master` if you're not already up to date
- [ ] Add or modify an example if there are changes to the public API
- [ ] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, **Changed**, etc)
- [ ] Run `rustfmt` on the project with `cargo fmt --all` - CI will not pass without this step
- [ ] Check that your branch is up to date with master and that CI is passing once the PR [... cut]
closed 2020-06-23 by jamwaffles