rust-embedded-community/ssd1306 #134: Will this driver also work with stm32f7xx-hal?

osphost, 2020-09-14:
Hey, I'm new to rust and wondering if this driver can also work with the `stm32f7xx-hal`  https://github.com/stm32-rs/stm32f7xx-hal/

It exposes a similar `BlockingI2c` struct: https://github.com/stm32-rs/stm32f7xx-hal/blob/471ca4de7026612b28f419b10f0ec0ef77813071/src/i2c.rs but I'm struggling to glue things together.

Thanks for any assistance

therealprof, 2020-09-14:
> but I'm struggling to glue things together.

What's the problem? Should be rather straight forward and similar to other STM32 code, including the examples in in this crate.

> To me it seems that the st32f7xx-hal isn't particularly stable and is a WIP.

I'd say it is rather stable especially since people are using it commercially.

> Maybe it would be better to not use it for now. Is it possible to use this library without it?

You will definitely need some I2C peripheral with a driver implementing the `embedded-hal` traits.

osphost, 2020-09-14:
Thanks for the quick reply.

I understand much clearer now, I didn't realize that `embedded-hal` traits are what interfaced the different specific device HALs (stm32f7xx, stm32f4xx etc) with your driver library.

I now have things compiling!

closed 2020-09-14 by osphost