rust-embedded-community/ssd1306 #135: Inconsistent Uppercase Font in Terminal Mode

Tenebryo, 2020-09-15:
- Version of `ssd1306` in use (if applicable): `0.4.0`
- MCU/other hardware in use: atmega328p
- Display resolution and interface: [I2C], [128x64]

## Description of the problem/feature request/other

Several uppercase characters in terminal mode are bold, inconsistent with the rest. The characters in question are `DLORW`, which use 2-3 pixel font instead of 1 pixel like the rest of the characters.

## Example code and result:

```rust
let interface = I2CDIBuilder::new().init(i2c);
let mut disp: TerminalMode<_> = Builder::new().connect(interface).into();

disp.init().unwrap();

disp.clear().unwrap();

for c in "!\"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\\]^_`abcdefghijklmnopqrstuvwxyz{|".chars() {
    disp.print_char(c);
}
```
[image]

jamwaffles, 2020-09-16:
In a sense this is by design; the font used renders those characters that way but I do agree it looks strange. We could certainly switch to a different font that's more consistent though. @therealprof wrote the initial terminalmode impl, so he might have a reason the current font was chosen. Font suggestions are welcome. I'd be happy to write a lil conversion script for one of the fonts in embedded-graphics if there's no other preference.

therealprof, 2020-09-16:
No specific reason why I picked this font. Happy to see it replaced by a better one. Do we have a nice 7x7 (or smaller) font somewhere?

jamwaffles, 2020-12-01:
I've switched to a new font which looks like this: 

[image]

This will go out in the next release.

jamwaffles, 2020-12-01:
Released in [0.4.1](https://crates.io/crates/ssd1306/0.4.1).

commit 05b0d84e (2020-10-21): New, more consistent terminal font

This font is taken from the ATIx_550_8x8.png font image from the
romfonts repo. A bitmap can be found here:
https://github.com/spacerace/romfont/blob/master/font-images/ATIx550_8x8.png

I don't have hardware to test with tat the moment so I might have messed
up the character mappings. I'd appreciate if someone could test on a
physical display and see how this looks.

I'm not including any of the special characters defined in that font. If
we want to add them, I'd suggest we do that in a future PR.

Closes #135
PR #137 (2020-12-01): New, more consistent terminal font
Hi! Thank you for helping out with SSD1306 development! Please:

- [x] Check that you've added documentation to any new methods
- [x] Rebase from `master` if you're not already up to date
- [x] Add or modify an example if there are changes to the public API
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, **Changed**, etc)
- [x] Run `rustfmt` on the project with `cargo fmt --all` - CI will not pass without this step
- [x] Check that your branch is up to date with master and that CI is passing  [... cut]
closed 2020-12-01 by jamwaffles