rust-embedded-community/ssd1306 #136: How to change addressing mode?

nebelgrau77, 2020-10-20:
- Version of `ssd1306` in use: 0.4
- MCU/other hardware in use: nRF52840
- Display resolution and interface: I2C, 128x64

## How to use addressing mode?

I'm experimenting with sending images over serial to the board. I'm using the `.draw` method like in the white noise example,
sending the serial RX buffer to the display. I noticed that the bytes are written top to bottom, not in horizontal lines as I expected.
I understand this is the column mode: is it the default one? How can I change it to Horizontal, or in other words, how do I use `init_with_mode` or `change_mode`?

jamwaffles, 2020-10-21:
> I understand this is the the default page mode?

That's correct. It's been a while since I looked at the datasheet but IIRC there's no way to change this behaviour - the display only works with 8 pixel high pages. What order are the pixels from the serial buffer in? You'll probably have to do some transformations to turn them into the correct format to write to the raw display.

nebelgrau77, 2020-10-21:
Ok gotcha. At the moment they're in simple line-after-line order: 8 pixels, another 8 pixels, etc. until the end of the row, then anothe row and so on. I will need to change my code to make my bytes be in the expected order, then. No problem :) Thanks!

closed 2020-10-21 by nebelgrau77