rust-embedded-community/ssd1306 #138: wrong part of the screen is updated for Rotate180 and <128 wide screens

martin2250, 2020-10-24:
- Version of `ssd1306` in use (if applicable): 0.4.0
- MCU/other hardware in use: STM32F072CBT6
- Display resolution and interface: I2C, 96x16

Hi James,

I've run into an issue with setting the rotation of the OLED. It seems that for `DisplayRotation::Rotate180`, the wrong part of the display is updated (approx. 128-96=32 columns at the right side of the screen contain only random noise).
I don't have a good debugging setup here for fixing this, so I hope you can reproduce this bug and help me out :)

Cheers!
Martin

[image]

```rust
#![no_main]
#![no_std]

use panic_halt as _;
use cortex_m_rt::entry;

use stm32f0xx_hal as hal;

use crate::hal::{
    stm32,
    prelude::*,
    i2c::I2c,
};

use embedded_graphics::{
    fonts::{Font6x8, Text},
    pixelcolor::BinaryColor,
    prelude::*,
    style::{TextStyle, TextStyleBuilder},
};

use ssd1306::{prelude::*, Builder, I2CDIBuilder, displayrotation::DisplayRotation};

#[entry]
fn main() -> ! {
    let mut p = stm32::Peripherals::take().unwrap();
    let mut cp = cortex_m::Peripherals::take().unwrap();

    cortex_m::interrupt::free(move |cs| {
        let mut flash = p.FLASH;
        let mut rcc = p.RCC.configure().freeze(&mut flash);

        let gpioa = p.GPIOA.split(&mut rcc);
        let gpiob = p.GPIOB.split(&mut rcc);
        let gpioc = p.GPIOC.split(&mut rcc);

        let mut delay = hal::delay::Delay::new(cp.SYST, &rcc);
        // lcd reset pin
        let mut lcd_rst = gpioa.pa10.into_open_drain_output(cs);

        // Configure pins for I2C
        let scl = gpiob.pb6.into_alternate_af1(cs);
        let sda = gpiob.pb7.into_alternate_af1(cs);

        let mut i2c = I2c::i2c1(p.I2C1, (scl, sda), 100.khz(), &mut rcc);

        let interface = I2CDIBuilder::new().init(i2c);
        let mut disp: GraphicsMode<_, DisplaySize96x16> = Builder::new().size(DisplaySize96x16).connect(interface).into();
        disp.reset(&mut lcd_rst, &mut delay).unwrap();
        disp.init().unwrap();
        disp.set_rotation(DisplayRotation::Rotate180).unwrap();
        disp.flush().unwrap();
        disp.clear();
        disp.flush().unwrap();

        let text_style = TextStyleBuilder::new(Font6x8)
        .text_color(BinaryColor::On)
        .build();

        Text::new("Hello world!", Point::zero())
            .into_styled(text_style)
            .draw(&mut disp)
            .unwrap();

        disp.flush().unwrap();

        let btn = gpioa.pa3.into_pull_down_input(cs);
        loop {
            if btn.is_high().unwrap() {
                cortex_m::peripheral::SCB::sys_reset();
            }
        }
    });
    
    loop {}
}
```

jamwaffles, 2020-10-24:
Thanks for the bug report, especially with a test case attached! I'll try and reproduce this on my end when I have a few minutes spare.

martin2250, 2020-10-24:
Did some more tinkering, this fixed it (`src/mode/graphics.rs`):

[image]

Now this only needs to be selected automatically somehow

bugadani, 2020-12-02:
It would help if someone who has a 72x40 could figure out the offsets necessary for a fix.

PR #145 (2021-01-17): Fix display rotation for different screen sizes
Hi! Thank you for helping out with SSD1306 development! Please:

- [ ] Check that you've added documentation to any new methods
- [ ] Rebase from `master` if you're not already up to date
- [ ] Add or modify an example if there are changes to the public API
- [ ] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, **Changed**, etc)
- [ ] Run `rustfmt` on the project with `cargo fmt --all` - CI will not pass without this step
- [ ] Check that your branch is up to date with master and that C [... cut]
closed 2021-01-17 by jamwaffles