rust-embedded-community/ssd1306 #142: Breaking change released in patch release 0.4.2

eldruin, 2020-12-21:
- Version of `ssd1306` in use (if applicable): 0.4.2
- MCU/other hardware in use: stm32f103
- Display resolution and interface: [SPI/I2C], [128x64]

## Description

A breaking change (#139) was released as part of patch release 0.4.2.
Was this intended?

For the record, the error is similar to this:
```
error[E0107]: wrong number of type arguments: expected 2, found 1
  --> examples/mlx90615-temperature-display-bp.rs:82:19
   |
82 |     let mut disp: GraphicsMode<_> = Builder::new().connect(interface).into();
   |                   ^^^^^^^^^^^^^^^ expected 2 type arguments
```
See [here](https://github.com/eldruin/driver-examples/runs/1587593177)

jamwaffles, 2020-12-21:
My mistake! I must've been a bit hasty with the latest release. I've yanked 0.4.2 and released 0.5.0. This should allow 0.4.x to continue without breaking changes I think. Please close this issue if it's resolved on your end now.

eldruin, 2020-12-21:
That worked. Thank you for your work! :)

closed 2020-12-21 by eldruin

jamwaffles, 2020-12-21:
No worries. Apologies for the disruption :grimacing: