rust-embedded-community/ssd1306 #146: Nonblocking implementation

ostenning, 2021-01-19:
- MCU/other hardware in use: STM32h7
- Display resolution and interface: I2C, [128x64]

## Nonblocking I2C support

Hello, I'm attempting to build an application that utilizes this library in conjunction with doing real-time DAC/ADC functions. I have noticed that my DAC output, which is executed from a Timer producing a simple sine-wave is blocked by the I2C communication that this library executes, which distorts the output of the DAC.

Is there any chance that this library could be extended to support non-blocking interrupt driven I2C communication?
If there isn't too much involved (and perhaps some assistance), I could give it a shot.

Kind regards
Oliver

jamwaffles, 2021-02-27:
Sorry for the delay! @therealprof maintains the crate that `WriteOnlyDataCommand` so might be able to offer more insight, but yes I think it's enough to add a custom impl of `WriteOnlyDataCommand` and pass that into `Builder::new().connect(interface).into()` instead of the provided blocking implementations.

quentinmit, 2022-07-10:
There's an `embedded-hal-async` now that provides `async`/`await`-compatible bus interfaces. See https://github.com/jamwaffles/ssd1331/pull/13 for a PR adding support to the `ssd1311` crate and follow https://github.com/embedded-graphics/embedded-graphics/issues/622 for an upstream `embedded-graphics` trait.

bugadani, 2023-05-15:
cc https://github.com/jamwaffles/ssd1306/pull/178