rust-embedded-community/ssd1306 #164: ssd1306 with rtic 1.0.0

pdgilbert, 2022-02-07:
(some relationship with #162.)
- Version of `ssd1306` in use: recent git
- MCU/other hardware in use: `blackpill 411` using a recent git version of `stm32f4xx_hal` (`no-std`)
- Display resolution and interface:128x64  I2C `shared-bus` 

## Description of the problem/feature request/other
I am trying to use `ssd1306` with `rtic v1.0.0` and am stuck on two errors. One is a trait problem with the `draw` method trying to write on the display
<details>
<summary>Click to expand</summary>

```
 Compiling test-example v0.1.0 (/home/paul/githubClones/rust-integration-testing/dev-testing)
error[E0599]: the method `draw` exists for struct `Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>`, but its trait bounds were not satisfied
   --> examples/zzdisplay_stuff_rtic.rs:123:15
    |
123 |             ).draw(&mut cx.local.display).unwrap();
    |               ^^^^ method cannot be called on `Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>` due to unsatisfied trait bounds
    |
   ::: /home/paul/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-0.7.1/src/text/text.rs:23:1
    |
23  | pub struct Text<'a, S> {
    | ----------------------
    | |
    | doesn't satisfy `<_ as Iterator>::Item = Pixel<_>`
    | doesn't satisfy `_: Iterator`
    | doesn't satisfy `_: embedded_graphics::Drawable`
    | doesn't satisfy `_: embedded_graphics::iterator::PixelIteratorExt<_>`
    |
    = note: the following trait bounds were not satisfied:
            `<Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>> as Iterator>::Item = Pixel<_>`
            which is required by `Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::iterator::PixelIteratorExt<_>`
            `Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: Iterator`
            which is required by `Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::iterator::PixelIteratorExt<_>`
            `<&Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>> as Iterator>::Item = Pixel<_>`
            which is required by `&Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::iterator::PixelIteratorExt<_>`
            `&Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: Iterator`
            which is required by `&Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::iterator::PixelIteratorExt<_>`
            `&mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>: TextRenderer`
            which is required by `Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::Drawable`
            `<&mut Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>> as Iterator>::Item = Pixel<_>`
            which is required by `&mut Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::iterator::PixelIteratorExt<_>`
            `&mut Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: Iterator`
            which is required by `&mut Text<'_, &mut embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>>: embedded_graphics::iterator::PixelIteratorExt<_>`
```
</details>

The other is a problem with something not being thread safe.
<details>
<summary>Click to expand</summary>

```
error[E0277]: `(dyn GlyphMapping + 'static)` cannot be shared between threads safely
   --> examples/zzdisplay_stuff_rtic.rs:15:35
    |
15  | #[cfg_attr(feature = "stm32f4xx", app(device = stm32f4xx_hal::pac,   dispatchers = [TIM2, TIM3]))]
    |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `(dyn GlyphMapping + 'static)` cannot be shared between threads safely
    |
    = help: within `MonoFont<'static>`, the trait `Sync` is not implemented for `(dyn GlyphMapping + 'static)`
    = note: required because it appears within the type `&'static (dyn GlyphMapping + 'static)`
    = note: required because it appears within the type `MonoFont<'static>`
    = note: required because of the requirements on the impl of `Send` for `&'static MonoFont<'static>`
    = note: required because it appears within the type `embedded_graphics::mono_font::MonoTextStyle<'static, embedded_graphics::pixelcolor::BinaryColor>`
note: required by a bound in `assert_send`
   --> /home/paul/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-1.0.0/src/export.rs:107:8
    |
107 |     T: Send,
    |        ^^^^ required by this bound in `assert_send`
    = note: this error originates in the attribute macro `app` (in Nightly builds, run with -Z macro-backtrace for more info)

```
[... 165 lines cut ...]

#dwt-systick-monotonic = "1.0"   # this seems to be needed for adc_dma_rtic fiddling

embedded-hal = {version = "^0.2.4" }
#embedded-hal = { version = "1.0.0-alpha.6,<1.0.0-alpha.7", package = "embedded-hal" }  

embedded-graphics = ">=0.7"
#shared-bus = { version = "0.2.2", features = ["cortex-m"] } 
shared-bus = {  git = "https://github.com/Rahix/shared-bus", features = ["cortex-m"] } 
heapless = "0.7"
nb = ">=0.1.2"
libm = "0.2"
cortex-m = ">=0.7"
cortex-m-rt = ">=0.7.0"

cortex-m-semihosting = { version = "0.3.7" }
panic-semihosting    = { version = ">=0.5.2" }
panic-reset          = { version = ">=0.1.0" }
panic-halt           = { version = ">=0.2.0" }

[features]
stm32f4xx = ["stm32f4xx-hal/rt", "systick-monotonic"]
stm32f411     = ["stm32f4xx-hal/stm32f411"  ] 
```
</details>

jamwaffles, 2022-02-10:
Thanks for the detailed issue! I don't have much spare time at the moment but I'll look at reproducing your issue when I can :)

pdgilbert, 2022-11-07:
Thanks @afternoon for pointing out this workaround. I have now re-enabled ssd1306 code in several examples and all are compiling. I'll get around to testing on hardware sometime soon.

NikaOly, 2023-01-14:
** stanntxt**_[bj hd boo s in.  Bridhh h[@vi ribrienn B rove viwigg chh B rhh in r B boo[chh_h in b_]()]()]()_****

pdgilbert, 2023-12-08:
I have made this work with `RTIC v1` (`cortex-m-rtic = ">=1.0"`). I have several examples at https://github.com/pdgilbert/rust-integration-testing. See directory examples/. I have just started playing with` RTIC v2` but am running into problems compiling `rtic` with `embedded-hal 1.0.0-rc.2`. See https://github.com/rtic-rs/rtic/issues/862.

pdgilbert, 2023-12-17:
I have now compiled examples  using `RTIC v2` and `stm32f4xx_hal` with both `embedded-hal 1.0.0-rc2` and a fork using `rc3`.   These are in the `eh-1-rc2` and `eh-1-rc3` branches at https://github.com/pdgilbert/rust-integration-testing. (See the CI testing in Actions.) I have also run tested one example on hardware. This all uses the `stm32f4xx_hal` dual support for `embedded-hal 1.0.0` as well as an older version. That means user code and `rtic` can use `embedded-hal 1.0.0`  while `ssd1306` uses  `embedded-hal 0.2.7`. The `tree` for this is pretty ugly.  I think it must be considered a transition solution and I look forward to an `ssd1306` crate based on `embedded-hal 1.0.0` .

pdgilbert, 2023-12-17:
It seems likely that `rtic  v1.x.x` will be abandoned with the release of `embedded-hal 1.0.0`. Users really should convert to `rtic v2`. (This needs to be emphasized given this old issue title  "ssd1306 with rtic 1.0.0".)