rust-embedded-community/ssd1306 #167: Init fails with Adafrukt

afternoon, 2022-10-07:
- Version of `ssd1306` in use (if applicable): [version here]
- MCU/other hardware in use: [hardware description here]
- Display resolution and interface: [SPI/I2C], [128x64]

## Description of the problem/feature request/other

[description here]

## Test case (if applicable)

```rust
// Failing test case demonstrating the issue here
```

afternoon, 2022-10-07:
I managed to unearth that I'm getting a `display_interface::DisplayError::BusWriteError` when I call `init`.

rfuest, 2022-10-08:
The display might use a different I2C address. Try to change `new` into `new_alternate_address` in this line:
https://github.com/rp-rs/rp-hal/blob/8d52f37dd1056ad3dcef78c4a74b663771a8b952/boards/rp-pico/examples/pico_i2c_oled_display_ssd1306.rs#L140

afternoon, 2022-10-08:
> The display might use a different I2C address. Try to change `new` into `new_alternate_address` in this line:
> https://github.com/rp-rs/rp-hal/blob/8d52f37dd1056ad3dcef78c4a74b663771a8b952/boards/rp-pico/examples/pico_i2c_oled_display_ssd1306.rs#L140

Yep, that was the issue! Thank you for your help.

closed 2022-10-08 by afternoon