rust-embedded-community/ssd1306 #168: Noob Issue: Cant manage to get this building for the stm32f1xx_hal

zacck, 2022-10-20:
- Version of `ssd1306` in use (if applicable): [GME12864-42](https://www.alibaba.com/product-detail/Chinese-Small-Size-30pins-SPI-IIC_1600293937907.html#)

- MCU/other hardware in use: STM32f103
 
- Display resolution and interface: [SPI/I2C], [128x64]

## Description of the problem/feature request/other
My issue at the moment is that I can't get my source code to even build I run into a linker error when I try and declare the interface

My code
```
//! Print "Hello world!" with "Hello rust!" underneath. Uses the `embedded_graphics` crate to draw
//! the text with a 6x8 pixel font.
//!
//! This example is for the STM32F103 "Blue Pill" board using I2C1.
//!
//! Wiring connections are as follows for a CRIUS-branded display:
//!
//! ```
//!      Display -> Blue Pill
//! (black)  GND -> GND
//! (red)    +5V -> VCC
//! (yellow) SDA -> PB9
//! (green)  SCL -> PB8
//! ```
//!
//! Run on a Blue Pill with `cargo run --example text_i2c`.

#![no_std]
#![no_main]

use cortex_m_rt::{entry, exception, ExceptionFrame};
use embedded_graphics::{
    mono_font::{ascii::FONT_6X10, MonoTextStyleBuilder},
    pixelcolor::BinaryColor,
    prelude::*,
    text::{Baseline, Text},
};
use panic_halt as _;
use ssd1306::{prelude::*, I2CDisplayInterface, Ssd1306};
use stm32f1xx_hal::{
    i2c::{BlockingI2c, DutyCycle, Mode},
    prelude::*,
    stm32,
};

#[entry]
fn main() -> ! {
    let dp = stm32::Peripherals::take().unwrap();

    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();

    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let mut afio = dp.AFIO.constrain();

    let mut gpiob = dp.GPIOB.split();

    let scl = gpiob.pb8.into_alternate_open_drain(&mut gpiob.crh);
    let sda = gpiob.pb9.into_alternate_open_drain(&mut gpiob.crh);

    let i2c = BlockingI2c::i2c1(
        dp.I2C1,
        (scl, sda),
        &mut afio.mapr,
        Mode::Fast {
            frequency: 400.kHz(),
            duty_cycle: DutyCycle::Ratio2to1,
        },
[... 82 lines cut ...]
  = note: "rust-lld" "-flavor" "gnu" "/var/folders/ys/gyj5sv6x4cg9cwzjdpj_cw3r0000gn/T/rustcf3uEWJ/symbols.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.10yil6jvfxhxvoly.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.14rg6runaeuwgi7k.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.1600vbpxdlzrwaxc.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.16fhc1jljqvv2iu6.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.1e6kmwo5eq0p1koe.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.1et1trfp11lcga9x.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.1smok0bnzyrutori.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.1uqd41w13yuyxsde.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.1vglvnh2j6h44h8p.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.20yju4kyjd7bbtd2.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.22du24j83lri8bw2.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.231eymdfzodsn20x.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.24tt6o63nfdr31uh.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.267iv5ugm3it8gu8.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.29c1rc5xa1cvqykm.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.29ucdp0dbxef3atm.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2bp1k05tvcncwxd6.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2br78hv9lbpqlk7x.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2bxrf0emua8temb9.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2cr9t4t1qwfaus2i.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2dgob5cj6gt0bf40.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2ear8kurp4je1o5z.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2fuvere7u14ne3dw.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2hft6bi38pon4x1.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2j1tjbmnjjgd14kj.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2p45fcd4t55ms4kz.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.2r8q2zfz4uz23jb8.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.30msr3ynkl5qe4i8.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.325qjhrngn90mma7.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3aaak54im0mfpssq.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3cwxqxrhq06lxr5a.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3gyu8hw0b7o4ce7g.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3khl5q8xgknzgpsf.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3olfjc8bjhcw6evs.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3w4tscvgadmsrjxy.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3w8el6xvoypypdtu.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3y44shcw5y570pil.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3z93cfbylqgkvbr.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3za1bktn7oal1y5b.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.3znbpgb66v3hm2tl.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.40cty7wcdsugrezm.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.44kgpofot6awuelj.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.48b9ddn0m25bgf1t.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4b8d7sqs76wqf2fp.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4djrzprledj3qzvq.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4hlk8hgd50rgux00.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4iu14f6se8d5wdif.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4m23b9o6d5tuik5k.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4pa13fzhpohq643t.rcgu.o" "/Users/zacck/Documents/learn_rust/blinky/target/thumbv7m-none-eabi/debug/deps/blinky-014cac237aa8171c.4v8bmguo3q7wpvo [... cut ...]
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 602 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 612 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 648 bytes
          rust-lld: error: section '.text' will not fit in region 'FLASH': overflowed by 696 bytes
          rust-lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
```

thank you so much for your time!

jamwaffles, 2022-10-20:
You most likely need to run with `--release` - debug builds are often too large to fit in flash memory which is what the linker error is describing.

blaa, 2022-12-27:
Yes. I've just set this up with stm32f1xx_hal and --release is required pretty much all the time. Without using write!, semihosting, etc. it's possible to get this to fit without --release.

With --release I'm at 13.7k which leaves plenty of space still.

jamwaffles, 2022-12-27:
Great! Glad to hear it's working :)

closed 2022-12-27 by jamwaffles

blaa, 2022-12-27:
Just need to add that I in fact have sh1106 oled - that *almost* (which is a bit confusing) works with ssd1306. Text was a bit shifted, but otherwise would be usable. I was starting to try to fix memory offsets when I found some info that it might be a different chipset.

If you have a FAQ section somewhere it might be worth noting there (?).

Still this issue is certainly worth closing. ;)

jamwaffles, 2022-12-27:
Ah, in that case you might want to try the [sh1106 driver](https://crates.io/crates/sh1106) which should work better with that IC.

blaa, 2022-12-27:
Yeah, I did, works much better. Thanks for your work! My first time with embedded Rust and it seems great so far.

Cheers,