rust-embedded-community/ssd1306 #172: Static noise when using crate with embedded_graphics

Squirrelcoding, 2022-12-19:
- Version of `ssd1306` in use (if applicable): [0.7.1]
- MCU/other hardware in use: [arduino uno]
- Display resolution and interface: [I2C], [128x64]

## Description of the problem/feature request/other
The source code is linked at the bottom. So I'm trying to run the `graphics_i2c.rs` example and I tried adapting it to fit with my Arduino Uno. The code compiles and all, but the problem is that instead of displaying the shapes correctly I get some static noise:

[image]

Other examples that _don't_ use `embedded_graphics` do work such as the one on `README.md`. I'm not very experienced with microcontrollers but I suspect that the way I set up the connection may be the culprit.
```rs
let dp = arduino_hal::Peripherals::take().unwrap();
let pins = arduino_hal::pins!(dp);

let i2c = arduino_hal::I2c::new(
    dp.TWI,
    pins.a4.into_pull_up_input(),
    pins.a5.into_pull_up_input(),
    115200,
);
```
This is the only part of the code that differs from the example so it must be it! However, I don't know what's wrong with it. I checked the default baud rate of the ssd1306 and it's 115200, and that's the baud rate set. The appropriate connections seem to be set up correctly, so what is it? It can't be the hardware as the same screen worked just fine for other examples. It has a 5V connection.

[main.rs](https://gist.github.com/Squirrelcoding/41985f856d5529ba0cfcdc3b665f333f)
[Cargo.toml](https://gist.github.com/Squirrelcoding/f9ff7e141603fbec7f8d6e661d77df55)

jamwaffles, 2022-12-20:
115200 is quite strange for I2C. Can you try `100_000` (standard I2C rate I think) or maybe `400_000`?

You could also try adding a clear/flush just after init, e.g.

```rust
display.init().unwrap();
display.clear().unwrap();
display.flush().unwrap();
```

I've anecdotally had this fix the static issue for me in the past, but I'm still unsure of its root cause.

If possible, try a different SSD1306 module to check if it's a hardware fault. Some of mine just seem to die sometimes.

blaa, 2022-12-27:
Try SH1106 library. SSD1306 *ALMOST* works ok with screens using sh1106, but doesn't clear them ok, and in the terminal mode the text is shifted a bit (and possibly many other small differences).

Your screen on photo looks exactly like mine - sh1106. https://docs.rs/sh1106/latest/sh1106/#

papyDoctor, 2023-02-24:
I've exactly the same problem:
Version of ssd1306 in use (if applicable): [0.7.1]
MCU/other hardware in use: RPPico
Display resolution and interface: [I2C], [128x64]

Replacing this crate by the SH1106 leads to the same result:

[image]

papyDoctor, 2023-03-01:
Ok, my bad. That was a power-on bad reset.
That was subtle because the device was partially functioning (DisplayOn/Off,  Normal/Inverse, Contrast,...) but the commands related to the embedded display SRAM no. Then I presume the SRAM needs a well formed reset.
A lot of time lost for this :(