rust-embedded-community/ssd1306 #192: read or toggle pixel in bufferedMode?

thomas725, 2023-08-31:
- Version of `ssd1306` in use (if applicable): 0.8.1
- MCU/other hardware in use: ESP32
- Display resolution and interface: I2C, 128x32

I found I can use this method to set a pixel in the buffer:
https://github.com/jamwaffles/ssd1306/blob/5c50f1158546e02e8ddc1ad3889eaf8d25112053/src/mode/buffered_graphics.rs#L161

But I would now like to toggle a pixel without having to add another buffer to remember all the pixels previous state was, so I'd either need a toggle or a read method, but neither seems to exist.

I guess my description already hints onto the only solution: fork & add those functions myself? Or is there any easier way?

jamwaffles, 2023-08-31:
Hey, thanks for opening this. There's currently no way of getting a pixel value, but no reason there shouldn't be! I think a `pub fn pixel(&self, x: u32, y: u32) -> Option<bool>` method would be a good feature to add to `BufferedGraphics` if you would like to open a PR.

thomas725, 2023-08-31:
like that?
https://github.com/jamwaffles/ssd1306/pull/193

jamwaffles, 2023-08-31:
#193 looks good at first glance, but I'm not sure what you're after. Can you write out some (pseudo) code to better describe what's lacking in the current API vs what you'd like to see?

thomas725, 2023-08-31:
I'd like the ability to toggle pixels instead of just turning them either on or off.
Either directly (see https://github.com/thomas725/ssd1306/blob/a6cfe678bc95a55aa8a58367ea5338cd147aac9b/src/mode/buffered_graphics.rs#L204 ) or by enabling reading current pixel's value to then be able to set it to the opposite value manually.