rust-embedded-community/ssd1306 #200: trait `DrawTarget` is not implemented for ... / esp_idf_hal::i2c::*

MikeMitterer, 2023-10-14:
- Version of `ssd1306` in use (if applicable): [0.8.3]
- MCU/other hardware in use: [esp32 v1]
- Display resolution and interface: [I2C], [128x64]

## Description of the problem/feature request/other

I'm fairly new to Rust! For my experiments, I use a ESP32 DevKit with esp_idf_hal, and it's I2cDriver.
I try to stay with esp_idf_hal because of its widly used and tests SW-Stack.
However, if I compile the code below I get the following error message.

F... I have no idea how to implement  `DrawTarget` for `esp_idf_hal::i2c::I2cDriver``
Do I miss something or is it really necessary to implement the DrawTarget?

```
error[E0277]: the trait bound `Ssd1306<I2CInterface<esp_idf_hal::i2c::I2cDriver<'_>>, ssd1306::prelude::DisplaySize128x64, BufferedGraphicsMode<ssd1306::prelude::DisplaySize128x64>>: DrawTarget` is not satisfied
   --> src/main.rs:61:15
    |
61  |         .draw(&mut display)
    |          ---- ^^^^^^^^^^^^ the trait `DrawTarget` is not implemented for `Ssd1306<I2CInterface<esp_idf_hal::i2c::I2cDriver<'_>>, ssd1306::prelude::DisplaySize128x64, BufferedGraphicsMode<ssd1306::prelude::DisplaySize128x64>>`
    |          |
    |          required by a bound introduced by this call
    |
    = help: the following other types implement trait `DrawTarget`:
              mono_font::draw_target::MonoFontDrawTarget<'_, T, mono_font::draw_target::Foreground<<T as DrawTarget>::Color>>
              mono_font::draw_target::MonoFontDrawTarget<'_, T, mono_font::draw_target::Background<<T as DrawTarget>::Color>>
              mono_font::draw_target::MonoFontDrawTarget<'_, T, mono_font::draw_target::Both<<T as DrawTarget>::Color>>
              Clipped<'_, T>
              ColorConverted<'_, T, C>
              Cropped<'_, T>
              embedded_graphics::draw_target::Translated<'_, T>
              Framebuffer<C, RawU1, BO, WIDTH, HEIGHT, N>
            and 10 others
note: required by a bound in `embedded_graphics::Drawable::draw`
   --> /Users/mikemitterer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-graphics-core-0.4.0/src/drawable.rs:106:12
    |
104 |     fn draw<D>(&self, target: &mut D) -> Result<Self::Output, D::Error>
    |        ---- required by a bound in this associated function
105 |     where
106 |         D: DrawTarget<Color = Self::Color>;
    |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `Drawable::draw`

```

## Test case 

I also tried it with: (Similar error)
```rust
    let interface = I2CDisplayInterface::new(i2c);
    let mut display = Ssd1306::new(
        interface, DisplaySize128x64, DisplayRotation::Rotate0
    ).into_terminal_mode();
```

```rust
use esp_idf_sys as _; // If using the `binstart` feature of `esp-idf-sys`, always keep this module imported
use log::*;

use esp_idf_hal::delay::{FreeRtos, BLOCK};
use esp_idf_hal::gpio::*;
use esp_idf_hal::i2c::*;
use esp_idf_hal::peripherals::Peripherals;
use esp_idf_hal::prelude::*;
use ssd1306::{mode::BufferedGraphicsMode, prelude::*, I2CDisplayInterface, Ssd1306};
use embedded_graphics::{
    mono_font::{ascii::FONT_6X10, MonoTextStyleBuilder},
    pixelcolor::BinaryColor,
    prelude::*,
    text::{Baseline, Text},
};

const SSD1306_ADDRESS: u8 = 0x3c;

fn main()-> anyhow::Result<()> {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    info!("Hello, world - 3!");

    let peripherals = Peripherals::take().unwrap();

    let i2c = peripherals.i2c0;
    let sda = peripherals.pins.gpio21;
    let scl = peripherals.pins.gpio22;

    let config = I2cConfig::new().baudrate(100.kHz().into());
    let mut i2c = I2cDriver::new(i2c, sda, scl, &config)?;

    let interface = I2CDisplayInterface::new(i2c);
    let mut display = Ssd1306::new(
        interface,
        DisplaySize128x64,
        DisplayRotation::Rotate0,
    ).into_buffered_graphics_mode();

    display.init().unwrap();
    // let _ = display.clear();

    let text_style = MonoTextStyleBuilder::new()
        .font(&FONT_6X10)
        .text_color(BinaryColor::On)
        .build();

    let mut led = PinDriver::output(peripherals.pins.gpio14)?;

    // Pins 35,34,39,26
    let button = PinDriver::input(peripherals.pins.gpio35)?;

    led.set_low()?;

    Text::with_baseline("Hello world!", Point::zero(), text_style, Baseline::Top)
        .draw(&mut display)
        .unwrap();
    display.flush().unwrap();

    loop {
        if button.is_high() {
            led.set_high()?;
        } else {
            led.set_low()?;
        }

        // we are sleeping here to make sure the watchdog isn't triggered
        FreeRtos::delay_ms(10);
    }
}

```

aujinm, 2025-01-05:
Hi today i try this crete , This problem I think maybe something change.
#### The Main.rs
```rust
// use embedded_graphics::{
//     mono_font::{ascii::FONT_6X10, MonoTextStyleBuilder},
//     pixelcolor::BinaryColor,
//     prelude::*,
//     text::{Baseline, Text},
// };// this for text-test
use embedded_graphics::{
    image::{Image, ImageRaw},
    pixelcolor::{BinaryColor, Rgb565},
    prelude::*,
}; //2
use esp_idf_svc::hal::{
    i2c::{I2cConfig, I2cDriver},
    prelude::Peripherals,
    units::Hertz,
};
use rand::prelude::*;
use ssd1306::{prelude::*, I2CDisplayInterface, Ssd1306};
use tinybmp::Bmp;
fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello, world!");
    let dp = Peripherals::take().unwrap();
    let i2c = dp.i2c1;
    let scl = dp.pins.gpio48;
    let sda = dp.pins.gpio47;
    let config = I2cConfig::new().baudrate(Hertz(1_000_000).into());
    let i2c = I2cDriver::new(i2c, sda, scl, &config).unwrap();

    log::info!("lets go");
    let interface = I2CDisplayInterface::new(i2c);
    let mut display = Ssd1306::new(interface, DisplaySize128x64, DisplayRotation::Rotate0)
        .into_buffered_graphics_mode();
    // display.init().unwrap(); //1 text-test not forget top  comment
    // log::info!("init done---------------------------------------------------");
    // let text_style = MonoTextStyleBuilder::new()
    //     .font(&FONT_6X10)
    //     .text_color(BinaryColor::On)
    //     .background_color(BinaryColor::Off)
    //     .build();
    // Text::with_baseline("Hello, world!", Point::zero(), text_style, Baseline::Top)
    //     .draw(&mut display)
    //     .unwrap();
    // log::info!("flush done---------------------------------------------------");
    //-------------------------------------------------bmp-------------------------------------------------
    // display.init().unwrap();
    // let bmp = Bmp::from_slice(include_bytes!("../picture/rust.bmp")).unwrap();
    // let im: Image<Bmp<Rgb565>> = Image::new(&bmp, Point::new(32, 0));
    // im.draw(&mut display.color_converted()).unwrap();
    //
    ////---------------------------------------Png to raw---------------------------------------------------------
    //let raw: ImageRaw<BinaryColor> = ImageRaw::new(include_bytes!("../picture/rust.raw"), 64);
    //let im = Image::new(&raw, Point::new(32, 0));
    //im.draw(&mut display).unwrap();
    //----------------------------------- random raw data
    let mut buf = [0x00u8; 1024];
    let mut rng = SmallRng::seed_from_u64(0xdead_beef_cafe_d00d);
    loop {
        rng.fill_bytes(&mut buf);

        display.draw(&buf).unwrap();
    }
}
```

### The Cargo.toml

```toml

[dependencies]
log = "0.4.22"
esp-idf-svc = { version = "0.50", features = [
  "critical-section",
  "embassy-time-driver",
  "embassy-sync",
] }
ssd1306 = "0.9.0"
anyhow = "1.0.95"
embedded-graphics = "0.8.1"
tinybmp = "0.6.0"
rand = { version = "0.8.5", default-features = false, features = ["small_rng"] }
[build-dependencies]
embuild = "0.33"

```

Dont foget picture if you want show picture(raw or bmp) 
[image]

aujinm, 2025-01-05:
> Hi today i try this crete , This problem I think maybe something change.
> #### The Main.rs
> 
> ```rust
> // use embedded_graphics::{
> //     mono_font::{ascii::FONT_6X10, MonoTextStyleBuilder},
> //     pixelcolor::BinaryColor,
> //     prelude::*,
> //     text::{Baseline, Text},
> // };// this for text-test
> use embedded_graphics::{
>     image::{Image, ImageRaw},
>     pixelcolor::{BinaryColor, Rgb565},
>     prelude::*,
> }; //2
> use esp_idf_svc::hal::{
>     i2c::{I2cConfig, I2cDriver},
>     prelude::Peripherals,
>     units::Hertz,
> };
> use rand::prelude::*;
> use ssd1306::{prelude::*, I2CDisplayInterface, Ssd1306};
> use tinybmp::Bmp;
> fn main() {
>     esp_idf_svc::sys::link_patches();
>     esp_idf_svc::log::EspLogger::initialize_default();
> 
>     log::info!("Hello, world!");
>     let dp = Peripherals::take().unwrap();
>     let i2c = dp.i2c1;
>     let scl = dp.pins.gpio48;
>     let sda = dp.pins.gpio47;
>     let config = I2cConfig::new().baudrate(Hertz(1_000_000).into());
>     let i2c = I2cDriver::new(i2c, sda, scl, &config).unwrap();
> 
>     log::info!("lets go");
>     let interface = I2CDisplayInterface::new(i2c);
>     let mut display = Ssd1306::new(interface, DisplaySize128x64, DisplayRotation::Rotate0)
>         .into_buffered_graphics_mode();
>     // display.init().unwrap(); //1 text-test not forget top  comment
>     // log::info!("init done---------------------------------------------------");
>     // let text_style = MonoTextStyleBuilder::new()
>     //     .font(&FONT_6X10)
>     //     .text_color(BinaryColor::On)
>     //     .background_color(BinaryColor::Off)
>     //     .build();
>     // Text::with_baseline("Hello, world!", Point::zero(), text_style, Baseline::Top)
>     //     .draw(&mut display)
>     //     .unwrap();
>     // log::info!("flush done---------------------------------------------------");
>     //-------------------------------------------------bmp-------------------------------------------------
>     // display.init().unwrap();
>     // let bmp = Bmp::from_slice(include_bytes!("../picture/rust.bmp")).unwrap();
>     // let im: Image<Bmp<Rgb565>> = Image::new(&bmp, Point::new(32, 0));
>     // im.draw(&mut display.color_converted()).unwrap();
>     //
>     ////---------------------------------------Png to raw---------------------------------------------------------
>     //let raw: ImageRaw<BinaryColor> = ImageRaw::new(include_bytes!("../picture/rust.raw"), 64);
>     //let im = Image::new(&raw, Point::new(32, 0));
>     //im.draw(&mut display).unwrap();
>     //----------------------------------- random raw data
>     let mut buf = [0x00u8; 1024];
>     let mut rng = SmallRng::seed_from_u64(0xdead_beef_cafe_d00d);
>     loop {
>         rng.fill_bytes(&mut buf);
> 
>         display.draw(&buf).unwrap();
>     }
> }
> ```
> 
> ### The Cargo.toml
> 
> ```toml
> [dependencies]
> log = "0.4.22"
> esp-idf-svc = { version = "0.50", features = [
>   "critical-section",
>   "embassy-time-driver",
>   "embassy-sync",
> ] }
> ssd1306 = "0.9.0"
> anyhow = "1.0.95"
> embedded-graphics = "0.8.1"
> tinybmp = "0.6.0"
> rand = { version = "0.8.5", default-features = false, features = ["small_rng"] }
> [build-dependencies]
> embuild = "0.33"
> ```
> 
> Dont foget picture if you want show picture(raw or bmp) ![图片](https://private-user-images.githubusercontent.com/53369562/400202876-567ff97d-b5ad-48f8-933b-b0585c0a8a6d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwODM1NjUsIm5iZiI6MTczNjA4MzI2NSwicGF0aCI6Ii81MzM2OTU2Mi80MDAyMDI4NzYtNTY3ZmY5N2QtYjVhZC00OGY4LTkzM2ItYjA1ODVjMGE4YTZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDEzMjEwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc0NmNkNjRjZGRhZDM1MTc3MjY0YzI3NjI5NTQ1MzM5Mzg1MTQ4MTA0NmMxNjdmMTEyMDc1Y2RmMjg0Mzk2YmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.OBK69dE-q1cc1FD4PzLC5K04Ll5xaJ5ozyDdrmUnbEo)

I test it and work perfectly