rust-embedded-community/ssd1306 #209: :rotating_light: Finding a new owner for this crate

jamwaffles, 2024-03-26:
I've been burned out on maintenance of this crate for a long time now, and it's not fair to those who use it, or would like to see something better under the crate name on crates.io.

Therefore, I'd like to find a new maintainer for this crate. My only request is that I be kept in `package.authors` in `Cargo.toml` and the current Git history is preserved. Beyond that, I'll be happy to move these crates to your username, or perhaps the https://github.com/embedded-graphics/ or https://github.com/rust-embedded/ organisations and giving you admin rights, depending on what people think is best.

sjoerdsimons, 2024-06-10:
ooi any luck thusfar or plans to pull the trigger of moving to  r-e-c ? (Asking as a potential contributor)

jamwaffles, 2024-06-11:
This crate should end up in r-e-c soonish :)

closed 2024-06-11 by jamwaffles