rust-embedded-community/ssd1306 #219: ssd 1306 does not compile because of maybe-async-cfg 0.2.5

kelko, 2024-11-14:
- Version of `ssd1306` in use (if applicable): 0.9.0 (7ed51935bdc6e9e976d3d9a747e768589eaea0cf26f483213bfededc947c8624)
- MCU/other hardware in use: Speed Studio Expansion Board with XIAMO SP2040
- Display resolution and interface: I2C, 128x64

## Description of the problem/feature request/other

I have a running local project with ssd1306 crate in version 0.9.0. It all worked fine, until today I wanted a second project with basically the same setup.
In the second project the ssd1306 crate just wouldn't compile:

```
~/Developer/Nova-Vergleich/Embedded/sp2040-playground git:[pagr]
cargo build
   Compiling pio-parser v0.2.2
   Compiling ssd1306 v0.9.0
error[E0277]: `Result<(), DisplayError>` is not a future
  --> /CENSORED/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ssd1306-0.9.0/src/size.rs:87:56
   |
87 | ... false).send(iface).await
   |                       -^^^^^
   |                       ||
   |                       |`Result<(), DisplayError>` is not a future
   |                       help: remove the `.await`
   |
   = help: the trait `Future` is not implemented for `Result<(), DisplayError>`, which is required by `Result<(), DisplayError>: IntoFuture`
   = note: Result<(), DisplayError> must be a future or must implement `IntoFuture` to be awaited
   = note: required for `Result<(), DisplayError>` to implement `IntoFuture`

error[E0277]: `Result<(), DisplayError>` is not a future
   --> /CENSORED/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ssd1306-0.9.0/src/size.rs:105:57
    |
105 | ... false).send(iface).await
    |                       -^^^^^
    |                       ||
    |                       |`Result<(), DisplayError>` is not a future
    |                       help: remove the `.await`
    |
    = help: the trait `Future` is not implemented for `Result<(), DisplayError>`, which is required by `Result<(), DisplayError>: IntoFuture`
    = note: Result<(), DisplayError> must be a future or must implement `IntoFuture` to be awaited
    = note: required for `Result<(), DisplayError>` to implement `IntoFuture`

error[E0277]: `Result<(), DisplayError>` is not a future
   --> /CENSORED/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ssd1306-0.9.0/src/size.rs:123:57
    |
123 | ... false).send(iface).await
    |                       -^^^^^
    |                       ||
    |                       |`Result<(), DisplayError>` is not a future
    |                       help: remove the `.await`
    |
    = help: the trait `Future` is not implemented for `Result<(), DisplayError>`, which is required by `Result<(), DisplayError>: IntoFuture`
    = note: Result<(), DisplayError> must be a future or must implement `IntoFuture` to be awaited
    = note: required for `Result<(), DisplayError>` to implement `IntoFuture`

error[E0277]: `Result<(), DisplayError>` is not a future
   --> /CENSORED/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ssd1306-0.9.0/src/size.rs:143:56
    |
143 | ... false).send(iface).await?;
    |                       -^^^^^
    |                       ||
    |                       |`Result<(), DisplayError>` is not a future
    |                       help: remove the `.await`
    |
    = help: the trait `Future` is not implemented for `Result<(), DisplayError>`, which is required by `Result<(), DisplayError>: IntoFuture`
    = note: Result<(), DisplayError> must be a future or must implement `IntoFuture` to be awaited
    = note: required for `Result<(), DisplayError>` to implement `IntoFuture`

error[E0277]: `Result<(), DisplayError>` is not a future
   --> /CENSORED/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ssd1306-0.9.0/src/size.rs:144:55
    |
144 | ..., true).send(iface).await
[... 224 lines cut ...]
 dependencies = [
- "proc-macro-error",
+ "manyhow",
  "proc-macro2",
  "pulldown-cmark",
  "quote",
@@ -1178,6 +1202,17 @@ dependencies = [
  "version_check",
 ]

+[[package]]
+name = "proc-macro-utils"
+version = "0.10.0"
+source = "registry+https://github.com/rust-lang/crates.io-index"
+checksum = "eeaf08a13de400bc215877b5bdc088f241b12eb42f0a548d3390dc1c56bb7071"
+dependencies = [
+ "proc-macro2",
+ "quote",
+ "smallvec",
+]
+
 [[package]]
 name = "proc-macro2"
 version = "1.0.89"
```

Dicklessgreat, 2024-11-28:
I had the same issue, and solved this problem by:
```
[patch.crates-io]
ssd1306 = { git = "https://github.com/rust-embedded-community/ssd1306.git", rev = "f3a2f7aca421fbf3ddda45ecef0dfd1f0f12330e" }

```
at Cargo.toml
see also https://github.com/rust-embedded-community/ssd1306/pull/218

sjoerdsimons, 2025-02-22:
@eldruin any chance of a new release so that the released crate compiles again with newer maybe-async-cfg :) thanks!